qualcomm/eld #15: unittest failed on EmitRelocs

lsahn-gh, 2025-04-11:
```
Input file: <stdin>
Check file: /home/lsahn/projects/linkers/llvm-project/eld/test/AArch64/standalone/EmitRelocs/EmitRelocs.test

-dump-input=help explains the following input dump.

Input was:
<<<<<<
          1:  
          2: Relocation section '.relamycode_1' at offset 0x20a8 contains 2 entries: 
          3:  Offset Info Type Symbol's Value Symbol's Name + Addend 
          4: 0000000000001038 0000001500000113 R_AARCH64_ADR_PREL_PG_HI21 0000000000002004 x + 0 
          5: 000000000000103c 0000001500000115 R_AARCH64_ADD_ABS_LO12_NC 0000000000002004 x + 0 
          6:  
          7: Relocation section '.relamycode_2' at offset 0x20d8 contains 8 entries: 
          8:  Offset Info Type Symbol's Value Symbol's Name + Addend 
          9: 000000000000105c 0000001400000113 R_AARCH64_ADR_PREL_PG_HI21 0000000000002000 y + 0 
         10: 0000000000001060 0000001400000115 R_AARCH64_ADD_ABS_LO12_NC 0000000000002000 y + 0 
         11: 0000000000001068 000000130000011b R_AARCH64_CALL26 0000000000001078 bar + 0 
         12: 000000000000106c 000000100000011b R_AARCH64_CALL26 0000000000001000 bar2 + 0 
         13: 0000000000001078 0000001600000113 R_AARCH64_ADR_PREL_PG_HI21 0000000000002008 z + 0 
         14: 000000000000107c 0000001600000115 R_AARCH64_ADD_ABS_LO12_NC 0000000000002008 z + 0 
         15: 0000000000001088 0000001600000113 R_AARCH64_ADR_PREL_PG_HI21 0000000000002008 z + 0 
         16: 000000000000108c 0000001600000115 R_AARCH64_ADD_ABS_LO12_NC 0000000000002008 z + 0 
         17:  
         18: Relocation section '.rela.text' at offset 0x2198 contains 6 entries: 
         19:  Offset Info Type Symbol's Value Symbol's Name + Addend 
         20: 0000000000001000 0000001500000113 R_AARCH64_ADR_PREL_PG_HI21 0000000000002004 x + 0 
         21: 0000000000001004 0000001500000115 R_AARCH64_ADD_ABS_LO12_NC 0000000000002004 x + 0 
         22: 000000000000100c 0000001400000113 R_AARCH64_ADR_PREL_PG_HI21 0000000000002000 y + 0 
         23: 0000000000001010 0000001400000115 R_AARCH64_ADD_ABS_LO12_NC 0000000000002000 y + 0 
         24: 000000000000101c 0000001600000113 R_AARCH64_ADR_PREL_PG_HI21 0000000000002008 z + 0 
         25: 0000000000001020 0000001600000115 R_AARCH64_ADD_ABS_LO12_NC 0000000000002008 z + 0 
check:51                                                                                       X error: no match found
>>>>>>
```

The input was supposed to be ordered by `.rela.text`, `.relamycode_2`, and `.relamycode_1` respectively as defined in EmitRelocs.test but as the log, the order was totally random. In the case, the test would fail. It shouldn't be depended on the order.

clang version
```
clang version 18.1.8 (Fedora 18.1.8-2.fc40)
Target: x86_64-redhat-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Configuration file: /etc/clang/x86_64-redhat-linux-gnu-clang.cfg
```
llvm version : 20.1.2
eld version: main commit - 9d192ca29f0ffb7bc5017451348f83430b074360

androm3da, 2025-04-11:
> We are looking at this issue, and it looks like the test case needs to be updated when asserts are turned ON.

For now it might suffice to `# UNSUPPORTED: asserts` the test case?

lsahn-gh, 2025-04-11:
> This happens to fail because of asserts turned ON in the LLVM build. Can you turn off asserts and retry.
> 
> We are looking at this issue, and it looks like the test case needs to be updated when asserts are turned ON.

Thank you for the reply, I will try and report the result asap.

androm3da, 2025-04-11:
This underscores a limitation with `eld`'s test suite: it depends on programs like `clang` to generate inputs and those inputs may not be stable.  How far into the future/past is any given commit of `eld` compatible with `clang`?

Other llvm projects restrict their `lit` tests to running the program under test (and "minimal transformation" programs like binutils/objdump/readelf/yaml2obj/obj2yaml).

A longer term fix to avoid problems like these could be to redesign tests dependent on `clang` to take output from clang once and have that output be committed to the `eld` repo as a stable test input.

quic-seaswara, 2025-04-12:
@lsahn-gh  I am still looking into fixing the issue. Just giving you confidence to continue what you are doing and experimenting with, as this is not a functional issue.

lsahn-gh, 2025-04-12:
@quic-seaswara 

I just finished the unittest without failure by turning off the assertions you mentioned earlier and it seems working good. yet i am not sure what 'turning off assertions' would affect to the entire unittest though.

But I'd suggest the following approaches for people who wanna run unittest and get started eld project that:
1. modify README to guide to turn off the assertion by default (we still don't need to modify any source codes)
2. guide to turn it on only if any experts, who are into taking care of it, like new redesign, update or whatsoever.
3. however, as a maintainer, put this issue into a future milestone and keep tracking the issue.

the first options sounds good for me for now and I can take it only if it makes sense.

quic-seaswara, 2025-04-14:
For now changing the README makes sense. Let us go ahead and make asserts off.

lsahn-gh, 2025-04-17:
@quic-seaswara 

shoud we close this or do you want to keep opening this for tracking? because it seems nothing more since you've updated the README.

quic-seaswara, 2025-04-23:
Let us keep this open

PR #359 (2025-08-27): llvm-lit : use asserts mode for unsupported tests
A test fails randomly when asserts are turned ON, due to emit relocs non determinism and the relocation processing is multi-threaded.

Disable this test for now, and look into why the test would fail post 21.0 release

Fixes #15