qualcomm/eld #38: Proposal: standardize commit messages

lsahn-gh, 2025-04-16:
I'd suggest a commit message template:

### basic template
```
File (without extensions): short summary (72 chars or less, but not mandatory)

More detailed explanatory text. Wrap it to 72 characters. The blank
line separating the summary from the body is critical (unless you omit
the body entirely).

Write your commit message in the imperative: "Fix bug" and not "Fixed
bug" or "Fixes bug." This convention matches up with commit messages
generated by commands like git merge and git revert.

Further paragraphs come after blank lines.

- Bullet points are okay, too.
- Typically a hyphen or asterisk is used for the bullet, followed by a
  single space. Use a hanging indent.

```

Examples
```
DiagnosticEngine: sort the includes in alphabetical order inside Diag class

Clean up Diag class that to sort the includes in order to ....

```
or if changes from multiple files included in a single commit:
```
Optimize sorting performance in ABC

The current implementation has a bottlenet during XYZ.

```

BR

quic-seaswara, 2025-04-16:
@qualcomm/eld-maint let me know what you think.

I definitely would like to see a description of the commit, I do not think filenames would be useful in the commit message.

Developers who inspect the commit would anyway get to know what files are changed. 

I would leave it to the developer in those cases as well.

quic-seaswara, 2025-04-16:
@qualcomm/eld-maint  what do you think ?

lsahn-gh, 2025-04-16:
@quic-seaswara 

Thank you for your ideas.

> Developers who inspect the commit would anyway get to know what files are changed.

I prefer mentioning file or directory name which are involved in a commit like [1] and [2], because it's easy to guess which file involved in the commit by just looking at short summary for developers who aren't related to the commit. Therefore, they are easily sync-up or follow-up.

It's like giving more detail through short summary to people that didn't involve.

--
[1]: https://github.com/qualcomm/eld/commit/422e20cee019548f5ae910d50a21f32895ae0851
[2]: https://github.com/qualcomm/eld/commit/e60200ec777e9ef244f537864280c05ed7c343be

quic-seaswara, 2025-04-16:
It might become confusing if it involves multiple directories, it is very rare that you just change one file.

androm3da, 2025-04-16:
This commit message format described by @lsahn-gh sounds like what is in use in the kernel, QEMU projects.  IMO if you decide to use this, you are also encouraged/obligated to make the same "bite size" commits.  This would, in turn, require the project to either adopt merge commits, or to be patient/disciplined enough to rebase commit series.  And implicitly commits from a pull req should not get squashed.  

But if `eld` aspires to end up as an incubator project / and/or full adoption into the LLVM monorepo, then it's moot.  If that's the case, `eld` should adopt the commit idioms that LLVM uses.

quic-seaswara, 2025-04-16:
LLVM prefers a linear history and we would like to use linear history for `eld` as well. 

For commit messages, let us follow what LLVM does. 

https://llvm.org/docs/DeveloperPolicy.html#commit-messages

We would not like to enforce a format, but the commit should be descriptive, so that developers get a short summary of what the commit is all about. Developers coming from other projects can sink in their commit style as well  and we would accept patches as long as the commit message describes the overall commit.

I would like to note that, if the commit adds or changes behaviour, please document in the documentation files that we have been maintaining, if there is a need for users to have a faq as part of the commit, please add a FAQ as well.

partaror, 2025-04-16:
I agree with the above comments that we should not enforce a strict commit message style, and having directory name in the commit message also does not make sense for ELD. We prefer linear history (rebase strategy instead of merge) as LLVM.

lsahn-gh, 2025-04-17:
Great, no doubt we would like to take LLVM commit template. 

Let me close this issue for now but if someone (might) have another opinions, we could open this again later.

closed 2025-04-17 by lsahn-gh