qualcomm/eld #165: Revisit how the ELF OSABI field is set

jonathonpenix, 2025-06-19:
**Description**
Currently (for aarch64 targets at least), eld will set the ELF OSABI flag to ELFOSABI_LINUX/GNU if given a linux-y emulation or triple (`-m aarch64linux`, for example). Other linkers don't seem to do that--I'm not sure if lld will set ELFOSABI_LINUX in general, and my understanding from the links below is that ld looks for GNU-specific features.

The OSABI flag being set to ELFOSABI_LINUX in this way has caused some surprises in the past, so it might be worth changing this.

Checking for GNU-specific features or looking for ELFOSABI_LINUX/GNU in the "incoming" object files might be options worth considering (and both are discussed/mentioned in the links below).

A few maybe-helpful links:
- https://github.com/llvm/llvm-project/issues/59199 (related discussion in lld)
- https://github.com/llvm/llvm-project/issues/81740 (another related discussion in lld)
- https://github.com/llvm/llvm-project/issues/40541 (semi-related discussion in objcopy)

**Reproducer**
```
cat > empty.c << '!'
int main(int argc, char *argv[]) {
  return 0;
}
!

clang --target=aarch64-none-elf empty.c -c
ld.eld -m aarch64linux empty.o
# Shows: OS/ABI: UNIX - GNU
readelf -h a.out
```

quic-seaswara, 2025-06-19:
Instead of reading a GNU feature, I would like to choose the maximum OSABI from the list of compatible object files for that target.

For example, if you have object files with all these mixed OS ABI in the elf header, its hard to figure out what should be the output :

* ELFOSABI_NONE
* ELFOSABI_SYSV
* ELFOSABI_STANDALONE
* ELFOSABI_GNU

This will allow the implementation to be simple and easy to comprehend. 

Thoughts ?

jonathonpenix, 2025-06-20:
Poking at this a bit more, IIUC lld will take the first non-`ELFOSABI_NONE` it finds if it isn't given something explicitly?: https://github.com/llvm/llvm-project/blob/a05393a879b2950fccca66ff0e1b6c70c39838e4/lld/ELF/Driver.cpp#L2193-L2214

Seems lld also sets a few of the OSABI options based on the emulation: https://github.com/llvm/llvm-project/blob/a05393a879b2950fccca66ff0e1b6c70c39838e4/lld/ELF/Driver.cpp#L148-L193 But, I'm not sure how concerned we are about freebsd and the other targets there.

Maybe it makes sense to follow what lld did about first non-`ELFOSABI_NONE`? Could always warn or something if there were subsequent and different, non-`ELFOSABI_NONE` `ELFOSABI_*` types.

commit 50c322f9 (2025-07-25): Infer OS/ABI from input files

Infer the OS/ABI from input files instead from the emulation
specified.

Return the default ABI to be ELFOSABI_NONE to mimic lld

Resolves #165
PR #271 (2025-07-25): Infer OS/ABI from input files
Infer the OS/ABI from input files instead from the emulation specified.

Resolves #165
commit 0c8cb2a6 (2025-07-25): Infer OS/ABI from input files

Infer the OS/ABI from input files instead from the emulation
specified.

Return the default ABI to be ELFOSABI_NONE to mimic lld

Resolves #165
closed 2025-07-25 by quic-seaswara