qualcomm/eld #175: Relocation overflow errors when first PT_TLS segment is empty

jonathonpenix, 2025-06-21:
**Description**
eld is producing (I think) incorrect relocation overflow errors in the reproducer below.

I think the important condition is whether the first `PT_TLS` segment (`tls_init` below) is empty--in the example below, eld does **not** error if any of the below happen:
- The `tls_init` PHDR is omitted (and ex: `.tdata` is commented out)
- The empty `tls_init PT_TLS;` is moved after `tls PT_TLS;`
- There is data in `tls_init` (ex: creating a variable that ends up in `.tdata`)

This was seen when building picolibc for aarch64 (not sure if this can happen with other targets).

**Reproducer**
```
cat > test.c << '!'
__thread int a;

int main(int argc, char *argv[]) {
  return a;
}
!

cat > script.t << '!'
PHDRS {
  flash PT_LOAD;
  ram_init PT_LOAD;
  ram PT_LOAD;
  tls_init PT_TLS;
  tls PT_TLS;
}

MEMORY {
  flash (rx!w) : ORIGIN = 0x40000000, LENGTH = 0x100
  ram (w!rx)   : ORIGIN = 0x40000200, LENGTH = 0x100
}

SECTIONS {
  .text : { *(.text) } >flash AT>flash :flash
  .tdata : { *(.tdata) } >ram AT>flash :tls_init :ram_init
  .tbss (NOLOAD) : { *(.tbss) } >ram AT>ram :tls :ram
}
!

clang --target=aarch64-none-elf -c test.c
# ld and lld seem ok with this, though lld's output looks a bit odd to me
ld.lld -m aarch64elf test.o -T script.t
aarch64-linux-gnu-ld.bfd -m aarch64elf test.o -T script.t

# eld errors: 
# Error: Relocation overflow when applying relocation `R_AARCH64_TLSLE_ADD_TPREL_HI12' for symbol `a' referred from test.o[.text] symbol defined in test.o[.tbss]
# Error: Relocation overflow when applying relocation `R_AARCH64_TLSLE_ADD_TPREL_LO12_NC' for symbol `a' referred from test.o[.text] symbol defined in test.o[.tbss]
ld.eld -m aarch64elf test.o -T script.t
```

jonathonpenix, 2025-06-21:
Also, FWIW, I think the separate `PT_TLS` segments were introduced in https://github.com/picolibc/picolibc/commit/fe5191e347427a00e0b5aba34e002f914e6e5ad8

And, on the lld front, I'm not sure what the bug is/was, but apparently they were seeing some bug in lld with multiple `PT_TLS` segments? See https://github.com/picolibc/picolibc/commit/efeba4717cfdd4351d1efcb4eb77a51e29cc2250

quic-seaswara, 2025-06-23:
Interesting issue, the symbol values for the TLS symbols are defined as offsets from the TLS image. So having two segments just complicates things a bit.

We need to check what the GNU linker does here. 

Usually the number of TLS variables is small and I am not sure what this enables and curious to find more information.

Probably reaching out to picolibc community would be useful.

quic-seaswara, 2025-06-23:
I think lld should use the start address to be the base of the tbss section as well. I am not sure if there is some assumption here.

commit 49611894 (2025-08-15): add multiple tls segments support

Though its not common with standard loaders that support multiple TLS
segments, picolibc introduces multiple TLS segments. This is supported
on GNU linker and lld, so support this with eld as well.

Resolves #175