qualcomm/eld #311: eld does not align input section properly when the output section has explicit address in the linker script

parth-07, 2025-08-08:
eld does not align input section properly when the output section has explicit address in the linker script.

For example:

```bash
#!/usr/bin/env bash

cat >1.c <<\EOF
int a;
__attribute__((aligned(0x100)))
int b = 11;
int foo() { return 11; }
EOF

cat >script.t <<\EOF
PHDRS {
  A PT_LOAD;
}

SECTIONS {
  .data (0x3010) : { *(.data*) } :A 
  .bss : { *(.bss*) } :A 
}
EOF

LDs=("ld.eld -m elf64lriscv")
SFs=(eld)

clang -o 1.o -target riscv64 1.c -c 
for i in "${!SFs[@]}"; do
  ${LDs[$i]} -o 1.${SFs[$i]}.out 1.o -T script.t -Map 1.${SFs[$i]}.map.txt
done
```

From the map-file:

```
.data   0x3010  0x4 # Offset: 0x4010, LMA: 0x3100, Alignment: 0x100, Flags: SHF_ALLOC|SHF_WRITE, Type: SHT_PROGBITS, Segments : [ A ]
*(.data*) #Rule 1, script.t [1:0]
.data   0x3010  0x4     1.o     #SHT_PROGBITS,SHF_ALLOC|SHF_WRITE,256 <----- .data has incorrectly aligned address!
        0x3010          $d
        0x3010          b
*(.data) #Rule 2, Internal-LinkerScript (Implicit rule inserted by Linker) [0:0]
```

**.data   0x3010  0x4     1.o     #SHT_PROGBITS,SHF_ALLOC|SHF_WRITE,256**

`.data` has alignment of 256, but the address is unaligned.

quic-seaswara, 2025-08-08:
This behavior is same with bfd as well.

There are users who rely on this feature.

quic-seaswara, 2025-08-08:
```
$ ld.lld -m elf64lriscv 1.o -T script.t
ld.lld: warning: address (0x3010) of section .data is not a multiple of alignment (256)

$ readelf -S -W a.out | grep data
  [ 2] .data             PROGBITS        0000000000003010 004010 0000f4 00  WA  0   0 16
```

```
$ riscv-ld -m elf64lriscv 1.o -T script.t
riscv-ld: warning: a.out has a LOAD segment with RWX permissions

$ readelf -S -W a.out | grep data
  [ 2] .data             PROGBITS        0000000000003010 004010 0000f4 00  WA  0   0 16
```

I think we can emit a warning here as what `lld` does with -Wlinker-script

parth-07, 2025-08-08:
> This behavior is same with bfd as well.

I am observing a different behavior with bfd. bfd is adding a padding at the beginning of the section so that the input section address is satisfying input section alignment. This is the bfd map-file for the reproducer specified in the issue:

```
.data           0x0000000000003010       0xf4
 *(.data*)
 *fill*         0x0000000000003010       0xf0
 .data          0x0000000000003100        0x4 1.o
                0x0000000000003100                b
```